OpenStickCommunity/GP2040-CE #4: Reverse Mode Simultaneous Directional Inputs

troytroy22, 2022-12-11:
Reverse Causing Simultaneal DP Inputs And LS RS Inputs at same time In respective stick modes. 

To recreate this issue - Enable Reverse mode in webcofig, restart in controller mode, change the Stick mode to LS or RS. This makes Both DP and LS RS enter inputs at the same time.

TheTrainGoes, 2022-12-17:
We took a brief look into this, it is a conflict caused by the two addons being enabled at the same time due to how they are implemented.  

We are discussing the overall issue of conflicting addons as they are independently made and compatibility among all all other addons will become more challenging as more are rolled out.

For this one specifically we will see if there is a quick fix, otherwise it may just be listed as 'not compatible with the LS/DP/RS' setting.

TheTrainGoes, 2023-03-26:
This was fixed via combination of code cleanup and addressing some oddities.  Thank you for the report @troytroy22 and for the fix @arntsonl

closed 2023-03-26 by TheTrainGoes