OpenStickCommunity/GP2040-CE #6: RPi Pico LED

FreddoUK, 2022-12-14:
It would be useful for diagnostic purposes if the LED on the pico could be used as an indicator light.

A few ideas:

- Fast Blink if power only and no USB data connection
- Solid if power+data
- Flicker on button press
- Slow Blink if in web-config mode

TheTrainGoes, 2022-12-17:
Hi @FreddoUK ,

This is a great idea and we will look into implementation.

TheTrainGoes, 2022-12-17:
I am going to add this to the official bounty board and sponsor it.

Bounty number 7:  Enable onboard LED for visual status indicator

Web-UI element that has the following options:
1 - Off
2 - Mode indicator
3 - Input test mode

Off - LED is off
Mode indicator - LED is solid if unit is powered with connected data, LED blinks rapidly if powered with no data, LED blinks slowly when in web-config mode
Input test mode - LED is off but turns on when any input is pressed (LED turns off when button is released)

arntsonl, 2022-12-20:
This is implemented and merged into main.

closed 2022-12-20 by arntsonl