OpenStickCommunity/GP2040-CE #15: Force SH1106/"FORCE" mode in the display settings

TheTrainGoes, 2022-12-21:
Add an option to force the SSH1106 / LCD options in the display settings. This will solve the edge case with certain SH1106 displays not showing our OLED code.

@thesuccubus current main fork supports this but should be changed:
https://github.com/thesuccubus/GP2040-CE

mthiesen, 2023-01-04:
It might actually be possible to auto-detect the display type. [FlashFloppy](https://github.com/keirf/flashfloppy/) does this. Look at the [oled_probe_model](https://github.com/keirf/flashfloppy/blob/feb0b883da32552035ce5292539d76162329c682/src/display/lcd_stm32f105.c#L1002) function.

thesuccubus, 2023-01-06:
SSH1106 screens sometimes misidentify themselves and cause issues as they are actually 132 px high and try to mask themselves as an SSD display, in my fix I have to force it to recognize it. But having the default iAddr set to -1 (auto recognition) might catch more display types and solve other issues.

mthiesen, 2023-01-06:
Sure, it is just that currently it seems like there is not even an attempt made to auto-identify the display. This is why I think it might be worth giving the FlashFloppy approach a try. In the end there should be a way to set the display type explicitly of course, to catch cases where this fails.

thesuccubus, 2023-01-14:
The library already being used already tries auto identification when the default iAddr is set to -1, it does already work if you set this variable manually. In one of my test forks I do use that. So just one change on the iAddr should work

mthiesen, 2023-01-14:
This does not seem to work, at least not with my SH1106, I still get the same horizontal offset in the display.

Looking at the code of `odbI2CInit()` one can see that passing `-1` as `iAddr` triggers the I2C address detection code in line 462-468. But beyond that `iAddr` is not even used during initialization, in particular not to detect a display type.

Irrespective of the value of `iAddr` the code does make an attempt to differentiate between SH1106, SH1107 and SSD1306 (lines 482-496). I have just stepped through this code with the picoprobe debugger and verified that it does not detect my display correctly. In fact it does not take any of the `if` branches.

The only thing that works for me right now passing `OLED_132x64` as `iType` to `obdI2CInit()`.

FlashFloppy seems to exploit the fact, that SH1106 does allow readback from the internal display memory over I2C, while SSD1306 does not. So they write some random bytes to the display and attempt to read them back, if it works the display is detected as an SH1106, otherwise it is assumed to be an SSD1306. I think this should work here as well.

arntsonl, 2023-02-08:
Fix should be good, we can re-open this if anyone sees an issue but this should cover our basis.

closed 2023-02-08 by arntsonl