OpenStickCommunity/GP2040-CE #16: Add keyboard mode

TheTrainGoes, 2022-12-21:
"Same as HayBox. Code for the TinyUSB keyboard implementation is already in Hay, not sure if Crane's uses TUSB.
They take the approach of registering a composite device that supports both KB and gamepad usage.
Button mapping in webconfig."

xgpt, 2023-01-24:
I would love to see this functionality be added. The ability to dynamically configure keyboard input with this web configurator would be *gamechanging* for me and my use cases, specifically it would open doors for disabled gamers/computer users to better utilize this project for accessibility purposes and not just general hobbyist/leisure purposes.

CountParadox, 2023-02-28:
I agree this would be very useful.

@xgpt  you could look at QMK / Vial in the meantime for accessibility / remapping buttons to different keyboard keys on the fly.

xgpt, 2023-03-02:
That does look like the way I'm leaning. CircuitPython has been very fun however I believe I'm starting to run into bottlenecks because of interpreted vs. compiled performance with some edge case projects I'm working on.

xgpt, 2023-03-02:
Live configuration of my projects with just a text editor has been hugely helpful with my workflow. Suppose I'll have to give a little of that up though. Thanks for weighing in friend!

xgpt, 2023-06-07:
Awesome!