OpenStickCommunity/GP2040-CE #23: Bug: Documentation's link to the configs folder is broken

micfok, 2022-12-21:
From: https://gp2040-ce.info/#/?id=installation

Clicking on the link for the config folder leads to a 404 (https://gp2040-ce.info/#/configs)

[image]

[image]

TheTrainGoes, 2022-12-21:
Thanks for the heads up @micfok , 

Let me get this fixed up

TheTrainGoes, 2022-12-21:
Closing this out.  

@micfok this was a legacy thing where there used to be configs available that were not pre-compiled in the original GP2040 repo.  

Currently we have releases available for all of the supported configurations.  

If you are looking for something specific that is not on the list please let me know or send me a message on Discord an I can try to help out.

closed 2022-12-21 by TheTrainGoes

micfok, 2022-12-21:
Ah, you already helped me out last night when I was trying to find the default pin mapping for the Pi Pico. I had assumed that folder was where I was originally supposed to find that mapping because I wasn't able to find it on the documentation page. Figured I would raise that bug or highlight that the documentation doesn't currently direct newcomers what they should wire their first button to (in order to hit the `S2` to open the web configurator.)

TheTrainGoes, 2022-12-21:
Got it!  Glad we got it sorted.  I made a sec section of the documentation with all the shortcuts in one place to help.