OpenStickCommunity/GP2040-CE #26: After reconfiguring the button-to-pin mapping, the button shown on display does not update to the new position

ayaya2036, 2022-12-27:
When you press a remapped button, the display will show the button's original position instead of its new position.

TheTrainGoes, 2022-12-30:
Hey @ayaya2036,

The display is not dynamic based off of pin configuration.  We would have no way to know where a button is physically installed in your specific setup.

If you are making a permanent change to the layout of the buttons for your stick you can also do a custom compile of the firmware that has the layout as you want it.  

The specific layouts have the positional data for each of the buttons, you would just need to change that to display as you like.  

If you need any help with this please reach out on Discord and I will do my best to get a compile done for you.

closed 2022-12-30 by TheTrainGoes

ayaya2036, 2022-12-30:
Is it possible to have a "Default Layout" input field in the pin mapping page, so that we can know which pin refers to which button, such that the display can be dynamic based off the original and remapped pin configuration? Having a place to store the default layout will be convenient when remapping buttons too.

TheTrainGoes, 2022-12-31:
@ayaya2036 are you free to talk about this more on Discord?