OpenStickCommunity/GP2040-CE #31: Add support for GPIO Expanders

Fortinbra, 2023-01-01:
Adding support for the PCA9535 16bit GPIO expander from Texas Instruments
https://www.ti.com/document-viewer/PCA9535/datasheet/GUID-2E758BF2-C717-41D1-BC2E-80CE2E71C3C9

This will allow us to expand the number of digital pins by 16 per each expander added to the I²C bus.

schnoog, 2023-03-25:
Would be great if also MCP23017 support could be integrated.
These I2C port expanders include (switchable) pull up resistors.

n-i-x, 2023-05-28:
Initial PoC is at https://github.com/n-i-x/GP2040-CE/tree/i2c_expander

It supports PCA95xx expanders currently.

The branch needs to be updated to the new protobuf to continue progress.

Initial latency testing shows no major impact to latency.

TODO:

- Update branch to protobuf and migrate settings to it
- Convert pca object to a more generic expander interface to allow for additional expander types/protocols
- Support multiple expanders
- Support keyboard input (requires modifying the keyboard addon)
- Move addon init code out of this branch and into a more generic PR (I think there is another task for this)
- (long term) support operating on the same i2c bus as OLED displays

TheTrainGoes, 2024-06-11:
This work has been completed via #1025

closed 2024-06-11 by TheTrainGoes