OpenStickCommunity/GP2040-CE #35: Update input latency Google Spreadsheet on Readme

arntsonl, 2023-01-03:
The old link only goes to 0.4.3

https://docs.google.com/spreadsheets/d/1eeX0SCOYnUDZMYzt_69wDpjnB_XUtvsfvHJYxxgTj28/edit#gid=1559471406

Update this to a sheet owned by someone else, and go all the way up to 0.6.0

FeralAI, 2023-04-14:
> The old link only goes to 0.4.3
> 
> https://docs.google.com/spreadsheets/d/1eeX0SCOYnUDZMYzt_69wDpjnB_XUtvsfvHJYxxgTj28/edit#gid=1559471406
> 
> Update this to a sheet owned by someone else, and go all the way up to 0.6.0

Hey, someone else here! I'm working on an updated version of the spreadsheet for GP2040-CE. I added the final GP2040 v0.4.3 version as well for comparison. It's still in progress and testing will be finished on the next release. Heres's the link: https://docs.google.com/spreadsheets/d/1OLmVg04e2q4aY0tqSJuaHtD8YMNCQnP-MzzSTtQLVm4/edit?usp=sharing

FeralAI, 2023-05-06:
That's some pretty good feedback @BatiKING, thank you! I believe @Fortinbra was considering some new visualization for the latency tests, so this could be something to factor in for a revision to the data collection.

For now I think we can close this issue since:

* The spreadsheet link has been updated in the README
* There is current test data for the v0.7.0 release in the new sheet
* @arntsonl now also has access to edit this sheet for future test results

closed 2023-05-06 by FeralAI