OpenStickCommunity/GP2040-CE #38: [BUG?] in SWITCH + LS mode , left direction will out DP direction down

xboxgod, 2023-01-04:
in 0.53 ver if set to LS + switch mode on pc, up down right is no problem, 
but left  will same time out DP direction down

TheTrainGoes, 2023-01-04:
Hi @xboxgod,

I tested this on v0.6.0 and am unable to reproduce on either the Switch directly on in Switch mode on a PC (Windows 10).  

Can you please test on v0.6.0 and let me know if you can reproduce the same behaviour?

If you can, please include more information about the setup you are using.

deeebug, 2023-01-05:
There's some weird stuff going on with Steam input support turned on, perhaps try checking with that turned off.

xboxgod, 2023-01-10:
I didnt use Switch test this bug, I use Open Joystick Display test, and same problem on 
gamepad-tester.com

TheTrainGoes, 2023-01-10:
@xboxgod - to confirm, you have been able to replicate this on multiple versions of the firmware?

TheTrainGoes, 2023-01-13:
@xboxgod following up on this.  Does it only occur in Steam?

xboxgod, 2023-01-13:
Not actually a STEAM platform, I just only test mode with gamepad-tester.com & Open Joystick Display software.
Because other chips will not have such an error output.

xboxgod, 2023-01-15:
and I test again, 0.52 not have this problem.

0.53 have LS LEFT same time output DP DOWN and 0.60 ver is same problem.

TheTrainGoes, 2023-01-15:
Can you please post a video or screen capture of the issue?

xboxgod, 2023-01-19:
[image]

[image]

in 0.5.3 and 0.6 two of version SWITCH and DINPUT mode all have this problem , LS + LEFT direction will same time output DP DOWN;

but 0.5.2 all mode & 0.6 XINPUT is not have this problem. just like this:
[image]

troytroy22, 2023-01-21:
I could not replicate this issue even with both software shown with latest 0.6.0 firmware on Windows 10. Are multiple controllers showing up in Windows Game Controllers? Did you Flash Nuke The Pico before flashing of new firmware? I wonder if Dual Inputs are enabled or glitching from an old webconfig profile.

xboxgod, 2023-01-22:
I used windows 7 pro version, I didnt test one win11 or win10.
and I make sure just only one controller on pc
and every flash uf2 file, I am sure use nuke clearn uf2 file to clearn flash.

TheTrainGoes, 2023-01-22:
I am still not able to replicate this issue.  Do you have any other addons enabled?

TheTrainGoes, 2023-02-14:
I am going to close this as we have not been able to replicate.  

@xboxgod if you can replicate the issue on v0.6.1 please make a new issue and we can check it out.

xboxgod, 2023-02-25:
yes I tested on 0.6.1 Bug is gone, very nice, Thanks !

closed 2023-02-25 by xboxgod