OpenStickCommunity/GP2040-CE #49: can you add debounce settings ?

travmaa, 2023-01-13:
hello, can you add a separate debounce setting for each button to the web adjuster?
It can be assigned between 0-10ms.

TheTrainGoes, 2023-01-13:
Hi @travmaa ,

We spoke internally about this and at this time we are not interested in adding debounce settings to the project.  You are welcome to fork and add this on your own however.

Soundtoxin, 2023-11-08:
I tried playing a rhythm game that I'd previously used xinput mode (vivid/stasis) for with keyboard mode so that I could move around some of the buttons to my liking (mainly using down and right instead of down and left buttons for the left hand notes) and have it work for games without controller support as well (Quaver). Previously with xinput I was able to play the game quite well, but in keyboard mode I noticed I'm dropping some hold notes. My theory is it's double-pressing the letter occasionally. Not sure why I didn't have this issue using mostly the same physical buttons in xinput mode, but my first thought was that maybe a debounce setting could fix it. It seems debounce is off your radar for now, but I thought I'd add my experience here just in case. Maybe this is just a bug in keyboard mode also, or in the new firmware. I upgraded to 0.7.5 from 0.7.3 before trying keyboard mode because when I went into the web configurator it reminded me I was out of date, so I took the chance to upgrade.

DRFTRZero, 2023-11-08:
> in keyboard mode I noticed I'm dropping some hold notes. My theory is it's double-pressing the letter occasionally. Not sure why I didn't have this issue using mostly the same physical buttons in xinput mode. 

NKRO is not implemented in keyboard mode yet, so pressing multiple keys at the same time may result in some inputs being dropped. Debounce would not result in differing behavior across different input modes.

Soundtoxin, 2023-11-09:
@DRFTRZero Your theory seemed plausible enough, but I've now used a keyboard tester website and I'm able to press 7 buttons at once no problem. I also tried various combos of the buttons and taking fingers on and off the buttons. I did not observe any issues with key rollover. However if I go to a terminal and press the buttons, I notice duplicate letters/presses pretty often. I'm using a DIY build of a Flatbox rev 5 with hotswap sockets if it matters.