OpenStickCommunity/GP2040-CE #59: Question about GP2040-CE Functionality (True Analog Stick)

15wileyr, 2023-01-19:
Hello, I was reading through the documentation here. I've never written my own Raspberry Pi code, I've always used Arduino. I'm trying to make a custom encoder for an arcade cabinet that will use a true analog stick inputs with x input. I want to get real nitty gritty into the actual code of the encoder, like have switches that switch between 8 way vs 4 way vs 2 way vs true analog input for the left stick. I'm thinking I'm better off using the Arduino x input libraries as this GP2040 library is easier to set up but If I wanted to get into the deeper stuff like that then I would have to modify the library itself. For instance I can't find true analog input for the analog sticks with this library. Is true analog input possible? Is coding without the web interface using functions like "output left x input stick value +1035" possible or is this library built more to be plug and play?

The only Arduino downside is the USB+uploading code to it is janky haha which is why this library is appealing to me.

deeebug, 2023-01-19:
If I'm not mistaken, you can currently get analog inputs by modifying the following lines:
https://github.com/OpenStickCommunity/GP2040-CE/blob/main/configs/Pico/BoardConfig.h#L121

you'd set the GPIO pins 26 and 27 which support ADC on the Pico and connect it to your controller.

TheTrainGoes, 2023-01-19:
Hey @15wileyr 

If you are not already I would recommend joining our Discord to talk about this further.  

https://discord.gg/k2pxhke7q8

We currently have support for analogue as @deeebug said above.  As far as switching between 8 way / 4 way / 2 way there is a hardware and software component to this.

TheTrainGoes, 2023-01-19:
I am going to close this as not currently planned.

closed 2023-01-19 by TheTrainGoes