OpenStickCommunity/GP2040-CE #68: Documentation Request - Default Pins? Configs? Pictures already exist, website just needs links/embeds.

xgpt, 2023-01-24:
https://github.com/OpenStickCommunity/GP2040-CE/tree/main/configs

The configs directory is REQUIRED for someone who is building a controller for hand-wired or otherwise trying to utilize this software to figure out which pico pin does what thing

Please add this https://github.com/OpenStickCommunity/GP2040-CE/tree/main/configs/Pico picture to the website.

It's not very user friendly to dig through source to find pictures!

I feel like this is just broken links though? The Installation part of : https://gp2040-ce.info/#/ shows a list of links to these dead internal links 
 * https://gp2040-ce.info/#/configs/DURAL
 *  https://gp2040-ce.info/#/configs/DURAL
 * https://gp2040-ce.info/#/configs/FlatboxRev4
 * https://gp2040-ce.info/#/configs/WaveshareZero
 * https://gp2040-ce.info/#/configs/CrushCounter
 * https://gp2040-ce.info/#/configs/OSFRD

These links all show "404-not found"

but if they linked to even just the blobs internal to the project it would be very useful to people looking into the project, I can tell you all are trying to make this project inclusive by basing on platformIO and documenting how to jump in, this would be a good use of time for someone who has the time. Thanks!

:)

TheTrainGoes, 2023-01-24:
Hi @xgpt ,

This is a good call out.  

We would like people to use the default pinout wherever possible and not build board around the other community design pinouts.

I will get our main image added into the docs in a moment

TheTrainGoes, 2023-01-29:
This is fixed.

closed 2023-01-29 by TheTrainGoes