OpenStickCommunity/GP2040-CE #80: Web configuration not opening

Barrone123, 2023-02-04:
When I plug in the board, my computer detects it as a POKKEN CONTROLLER, and I can't seem to open the web configuration to change the mappings.
[image]
[image]

TheTrainGoes, 2023-02-04:
Hi @Barrone123 ,

This would indicate it is in Nintendo Switch mode.  

Have you read the documentation on changing controller modes by pressing a button on plug-in?

www.gp2040-ce.info

Barrone123, 2023-02-05:
Hello @TheTrainGoes , 

Yes, I've tried doing that, but it still does not work. However, I have already fixed the issue with the Pokken controller by nuking the pico and re-installing the code. Probably the worse way to do it, but it got the job done. 

I'll try again later, as I'm already using it to play games on my PC. Thank you for the comment!

TheTrainGoes, 2023-02-06:
Hi @Barrone123 ,

Please let me know if this is still an issue, otherwise I can close it.

Thanks!

TheTrainGoes, 2023-02-07:
Closing - will re-open if we can verify issue on most recent build.

closed 2023-02-08 by Fortinbra