OpenStickCommunity/GP2040-CE #107: Bounty - New LED mode - Individually set colors for on/off state via web-config

TheTrainGoes, 2023-02-19:
After some conversation in the BentoBox channel I am throwing a bounty up to see if we can get a new LED mode added that allows someone to configure a static on/off color per button / LED via web-config.

As an example, if I have a stickless layout I could make LEFT, DOWN, RIGHT, UP green and the rest red.  Then on press LEFT, DOWN, RIGHT, UP could turn blue while the rest could turn pink.

We already define colors in the code so those could be whats provided via the web-config.  

The web-config would have a section in the LED configuration to deal with this new mode.

FeralAI, 2023-04-14:
I'm starting work on this feature. Currently targeting the minimum viable implementation at the moment since there's a bigger LED refactor coming along.

TheTrainGoes, 2023-05-10:
This has been merged