OpenStickCommunity/GP2040-CE #115: fsdata.c missing from main branch

KonradMcClure, 2023-02-22:
Was trying set up my own configuration for a board I'm working on and ran into cmake issues due to main missing fsdata.c from the lib/httpd directory.

Not sure if that's intended for some reason but I went and grabbed it from another branch that looked fairly recent.

deeebug, 2023-02-22:
`fsdata.c` is no longer part of the repo source since it's a build output
It's setup to build itself but there might have been some dependency issues or some other cause.
Can you post what your CMake logs are?

Regardless, you can always grab the latest from one of these actions: https://github.com/OpenStickCommunity/GP2040-CE/actions/runs/4234231538

KonradMcClure, 2023-02-22:
Oh, okay that makes sense. I cloned the main branch again and tried to build and got the same result as before (I assume by CMake logs you mean the console output, unless there are different logs stored elsewhere):
```
[main] Configuring project: GP2040-CE 
[proc] Executing command: "C:\Program Files\cmake-3.20.1-windows-x86_64\bin\cmake.EXE" --no-warn-unused-cli -DCMAKE_EXPORT_COMPILE_COMMANDS:BOOL=TRUE -DCMAKE_BUILD_TYPE:STRING=Debug "-DCMAKE_C_COMPILER:FILEPATH=C:\Program Files (x86)\GNU Arm Embedded Toolchain\10 2020-q4-major\bin\arm-none-eabi-gcc.exe" "-DCMAKE_CXX_COMPILER:FILEPATH=C:\Program Files (x86)\GNU Arm Embedded Toolchain\10 2020-q4-major\bin\arm-none-eabi-g++.exe" -SD:/GITRepositories/Test/GP2040-CE -Bd:/GITRepositories/Test/GP2040-CE/build -G "NMake Makefiles"
[cmake] Not searching for unused variables given on the command line.
[cmake] PICO_SDK_PATH is D:/GITRepositories/Test/pico-sdk
[cmake] PICO platform is rp2040.
[main] Configuring project: GP2040-CE 
[cmake] Submodules updated
[cmake] -- Not Skipping WebBuild
[cmake] -- Could not find Node.js
[cmake] -- Could not find npm
[cmake] Build type is Debug
[cmake] Using regular optimized debug build (set PICO_DEOPTIMIZED_DEBUG=1 to de-optimize)
[cmake] Using PICO_BOARD from environment ('pico')
[cmake] Using board configuration from D:/GITRepositories/Test/pico-sdk/src/boards/include/boards/pico.h
[cmake] TinyUSB available at D:/GITRepositories/Test/pico-sdk/lib/tinyusb/src/portable/raspberrypi/rp2040; enabling build support for USB.
[cmake] Compiling TinyUSB with CFG_TUSB_DEBUG=1
[cmake] BTstack available at D:/GITRepositories/Test/pico-sdk/lib/btstack
[cmake] cyw43-driver available at D:/GITRepositories/Test/pico-sdk/lib/cyw43-driver
[cmake] Pico W Bluetooth build support available.
[cmake] lwIP available at D:/GITRepositories/Test/pico-sdk/lib/lwip
[cmake] mbedtls available at D:/GITRepositories/Test/pico-sdk/lib/mbedtls
[cmake] -- Configuring done
[cmake] CMake Error at lib/httpd/CMakeLists.txt:1 (add_library):
[cmake]   Cannot find source file:
[cmake] 
[cmake]     fsdata.c
[cmake] 
[cmake]   Tried extensions .c .C .c++ .cc .cpp .cxx .cu .mpp .m .M .mm .h .hh .h++
[cmake]   .hm .hpp .hxx .in .txx .f .F .for .f77 .f90 .f95 .f03 .ispc
[cmake] 
[cmake] 
[cmake] -- Generating done
[cmake] CMake Generate step failed.  Build files cannot be regenerated correctly.
[proc] The command: "C:\Program Files\cmake-3.20.1-windows-x86_64\bin\cmake.EXE" --no-warn-unused-cli -DCMAKE_EXPORT_COMPILE_COMMANDS:BOOL=TRUE -DCMAKE_BUILD_TYPE:STRING=Debug "-DCMAKE_C_COMPILER:FILEPATH=C:\Program Files (x86)\GNU Arm Embedded Toolchain\10 2020-q4-major\bin\arm-none-eabi-gcc.exe" "-DCMAKE_CXX_COMPILER:FILEPATH=C:\Program Files (x86)\GNU Arm Embedded Toolchain\10 2020-q4-major\bin\arm-none-eabi-g++.exe" -SD:/GITRepositories/Test/GP2040-CE -Bd:/GITRepositories/Test/GP2040-CE/build -G "NMake Makefiles" exited with code: 1
```

deeebug, 2023-02-22:
Yes, this is that exact log.

What you pasted has these lines:
```
[cmake] -- Could not find Node.js
[cmake] -- Could not find npm```

which means you don't have `node.js` installed. IIRC, you need version 17 installed and then it will build `fsdata.c`.

arntsonl, 2023-02-22:
If you want to download a premade fsData.c, you can check our github actions tab and grab one of the latest runs.

For example, https://github.com/OpenStickCommunity/GP2040-CE/actions/runs/4234231538 at the bottom of this is an fsData you can use if you want to run SKIP_WEBBUILD=true as @deeebug mentioned.

closed 2023-02-22 by arntsonl