OpenStickCommunity/GP2040-CE #118: Error versión 0.6.1

juannma001, 2023-02-28:
In the latest version there is an error when entering an example mvc2 game. Only the addresses work and the home button tested in version 0.6 and 0.6.1 on the home screen works perfectly. Greetings

TheTrainGoes, 2023-02-28:
Can you please tell us a bit more about this?   Does it work in other games or is it only MvC2?

juannma001, 2023-03-01:
Of course, I tried games like the kof xiii, ttt2, sf third strike, Mvc origins and in all of them it works perfectly, the error is in mvc2 in its three versions (use, EU, jp) it only allows you to enter the home screen by pressing start, but it only allows the movement if I activate the left analog with the d- It should be noted that if I install version 0.4.3 this error does not occur, I hope to be of help and thank you very much for all the hard work.

TheTrainGoes, 2023-03-01:
@juannma001  We did some significant work to address the PS3 home button issue in an earlier release.  

I don’t have a copy of this to test with so We will look into some potential fixes and I may reach out to you with test builds if you are up for it.  

Are you on Discord?

juannma001, 2023-03-01:
Of course, I will do my best to help. I joined the group this morning in discord

TheTrainGoes, 2023-03-26:
Closing the loop on this.  

Some testing was done on this and those changes impacted the function of the controller on the rest of the PS3 system.  

Given how legacy this is I would recommend using the version of GP2040/GP2040-CE that works for this specific game.

We are going to keep the PS3 code as it is with the fixed home button for PS3 native games.

closed 2023-03-26 by TheTrainGoes

TheTrainGoes, 2024-01-08:
Hi @juannma001 ,

I am reviving this very old thread with some interesting news.  

In doing some further testing on PS3 MvC2 I noticed that you can use PS4 mode on GP2040-CE and it works just fine in game but with no home button access.  

While maybe not perfectly idea, I think having something that works in-game is a goos balance point on this.  

If you are still around and playing with these, I would be interested to know if PS4 mode also works on the other titles you said had issues.