OpenStickCommunity/GP2040-CE #123: Bounty - Support for split/multi-piece controllers

sgfdevices, 2023-03-04:
I'd like to be able to support a split or multi-piece controller with GP2040-CE.  

Ideally, this would be accomplished by running two separate controllers with their own RP2040 and own copy of the firmware. Each one would use a different configuration file when built. Each configuration file would specify whether the device it's intended for is a main/host or sub/client controller. The host would be the one that plugs into the console/computer and reports the inputs. The client would only connect to computers for firmware updates, and would otherwise only send its button state to the host and receive LED data from the host.  

[This should also support Issue 101](https://github.com/OpenStickCommunity/GP2040-CE/issues/101) in such a way that the same gamepad button can be bound on both the host and client units, and the input would be triggered when either physical button is pressed.  

The comms between the host and client need to happen with a simple 2 wire protocol so minimal cabling over 2 GPIO pins can be used. For example, a TRRS cable like many split keyboards. Ideally UART can be used, but having support for more advanced protocols like RS485 would be great for future proofing if possible.

Caian, 2023-03-28:
I was looking into this topic, as I am trying to rebuild a fried Dual Shock 4 controller and ran out of ADC's on rp2040.

TheTrainGoes, 2023-09-06:
@sgfdevices we are going to close this one for now and do re-post once we do a little cleanup and reset.  Toss me a DM.

closed 2023-09-06 by TheTrainGoes