OpenStickCommunity/GP2040-CE #125: Unexpected behavior with an raspberry pico WIFI

darkomen, 2023-03-13:
I have loaded the GP2040-CE_0.6.2_Pico.uf2 binary on a Raspberry Pico WIFI without any board more and the controller stop working. After severals plugs and unplugs the wire the controller appears to work, but after a minute working the controller stop again.The system for gaming is a raspberry pi 3 with retropie. In the system logs, the controller appears to we installed correctly. I have to load again the firmware and the behaviour is the same as described. I have tested with a Raspberry Pico and the controller works fine. Is any issue with the pico wifi?

I can't know if the controller is ON, because I don't have a computer with windows and the web configuration doesn't work. is there any GPIO with a powerON led functionality?

TheTrainGoes, 2023-03-13:
Hi @darkomen ,

I have been hearing of Pico W based issues from a few similar projects.  As it stands GP2040-CE is designed for use on a Pico only and not a Pico W.  

We are looking into the Pico W to make a wireless proof of concept but that is a while out.

I have a Pico W so I will load up 0.6.2 when I have a chance and see if I can replicate this issue.

darkomen, 2023-03-14:
Thanks for your support, If you need some help from my site, don't hesitate to contact me.

Doy you know something about my question for the led? Or there is any document about the led support for the firmware?

Thanks

TheTrainGoes, 2023-03-14:
Hi @darkomen ,

Web-config should work just fine on Mac / PC / Linux now.  

There was an add-on built that allows use of the onboard LED from the Pico however the Pico W does not allow access to the LED in the same way. 

You may be able to modify the code and compile to test this, but again, it is currently unsupported.

darkomen, 2023-03-14:
I've tried to install de nuke_flash.uf2 firmware, but the controller still doesn't work :(

For the web-config, is necesary a specify configuration in computer? It doesn't work neither pico nor pico W.
Regards

TheTrainGoes, 2023-03-15:
Hi @darkomen are you on Discord at all in our server?

I have compiled a version of the firmware that removes all turbo and turbo LED functions which appears to work on the Pico W and would like to have you try it out.

darkomen, 2023-03-15:
Hi, @TheTrainGoes I'm not in discord but this afternoon I can join and we can test it! Thanks you!
Regards

TheTrainGoes, 2023-03-26:
Just wanted to close the loop on this.  

I did some testing with my Pico W and was not able to replicate the behaviour.  Having said that there are some default pins that should be changed for the Pico W.  

Overall I am going to say that the Pico W is not supported at this time but should work just fine with some pin assignment changes.  It's possible that you have a bad Pico W although that would require testing on another Pico W to confirm.

closed 2023-03-26 by TheTrainGoes