OpenStickCommunity/GP2040-CE #126: Please add a 4-way input method to the new firmware ↖↗↙↘

fangxiang-gan, 2023-03-18:
Please add a 4-way input method to the new firmware ↖↗↙↘， The default 16, 17, 18, and 19 pins are good preset schemes that truly implement 8-way input. This is very necessary for combat games and players, thank you.

StdVectorBool, 2023-03-18:
What combat games in particular need dedicated diagonal inputs?

fangxiang-gan, 2023-03-18:
Street Fighters and KOF all need it, especially Byson, Honda, and other brute force wrestlers. These four directions, plus the original up, down, left, and right directions, total 8 types.
↖ ↑ ↗
←    →
↙ ↓ ↘

fangxiang-gan, 2023-03-18:
Yes. a stickless controller with 8 directional buttons.

TheTrainGoes, 2023-03-18:
Hi @fangxiang-gan ,

What advantage would dedicated inputs have over double wired buttons?

This is the first request we have had for this, and we have a lot of Street Fighter and Tekken players.

fangxiang-gan, 2023-03-18:
Hi@TheTrainGoes 
For example, squatting and jumping in KOF ↙↗， Boxing champion among street bullies ↙ → Squatting and preventing back pressure accumulation can also be completed with one finger. "You don't need to occupy two fingers in two directions. Getting used to it can be very convenient in the future.". For example, in the horizontal version of the clear version game, oblique running can also be easily achieved, and the movement in various directions in the shooting game will be more comfortable.

TheTrainGoes, 2023-03-18:
Hi @fangxiang-gan ,

Why not just double wire to a second button for these instances then?  

Have a button for ↙ and have it wired to [LEFT + DOWN] while also having separate buttons for [LEFT] and [DOWN].

fangxiang-gan, 2023-03-18:
Hi@TheTrainGoes
In hardware, if two pins are connected to a button, a single command will be short-circuited. To be specific, for example, if the left and the down are connected together, when I want to trigger the left independently, the down will also conduct. Therefore, the hardware can only solve one-way conduction through a two-stage transistor bridge circuit. To achieve these four sets of directions, at least two sets of bridge circuits, namely eight diodes, are required, which is too costly.  So it would be great if it could be implemented in software.  Thank you very much~

TheTrainGoes, 2023-03-18:
Hi @fangxiang-gan ,

I am very familiar with how buttons work, thank you.

Are you willing to host a bounty for this feature?

fangxiang-gan, 2023-03-18:
Hi@TheTrainGoes
This is my first time using github so I don't know how to offer a reward. Is it integral? Virtual currency?

TheTrainGoes, 2023-03-18:
@fangxiang-gan I would recommend joining the Discord server.

fangxiang-gan, 2023-03-18:
@TheTrainGoes
Thank you for your reply. Thank you.

TheTrainGoes, 2023-03-26:
Going to close out this request.  

Have spoken with a few people in the FGC and don't see an immediate benefit for it.  May introduce complicated SOCD logic if implemented as well.  

I would recommend either the hardware based solution with diodes or doing a fork and implementing yourself in the code.

closed 2023-03-26 by TheTrainGoes