OpenStickCommunity/GP2040-CE #128: Is it normal that my up and down buttons have delayed inputs?

Barrone123, 2023-03-27:
my up and down buttons have noticeable input lags in-between presse, is that normal for the 0.6.2 update?

arntsonl, 2023-04-04:
It is not normal @Barrone123 , can you let us know what add-ons you have enabled as well as what hardware you are running on?

TheTrainGoes, 2023-04-06:
Hey @Barrone123 ,

Still waiting on a reply for this.  If you are on Discord feel free to hope onto the convo there.  If we don't hear back from you in a few days on this we will close the issue out.

https://discord.gg/k2pxhke7q8

Barrone123, 2023-04-07:
Hello, sorry for the late reply. As of today, the fix I did was to revert to 6.0 of the build. 6.2 on my Pi Pico controller has many problems:

First, the Up and Down buttons have a distinct lag between pressing them; it is especially present when playing as it limits movement options.
Second, Triple button combos like down, forward, + any button. Does not work at all, It only works with double buttons combos, and even then, it has specific timings. (My controller style is a Hitbox controller)

That is all that I found with the 6.2 build.

Barrone123, 2023-04-07:
> It is not normal @Barrone123 , can you let us know what add-ons you have enabled as well as what hardware you are running on?

If I remember correctly I did not add any add-ons on my build (like LED's and such) it is as minimalistic as it gets

TheTrainGoes, 2023-04-07:
Hi @Barrone123 ,

Can you provide details on what Pico device you are using, how you have it setup and also possibly some video recordings of the issue?   

If you are on the Discord we can chat there as well.

Barrone123, 2023-04-07:
> Hi @Barrone123 ,
> 
> Can you provide details on what Pico device you are using, how you have it setup and also possibly some video recordings of the issue?
> 
> If you are on the Discord we can chat there as well.

Okok, I'll Dm you right now. My Discord name is Denkiin#9749.

TheTrainGoes, 2023-05-15:
@Barrone123 ,

We believe this has been resolved in [v0.7.1-RC2](https://github.com/OpenStickCommunity/GP2040-CE/releases/tag/v0.7.1-RC2).  If you can check out the pre-release we have that would be great.

arntsonl, 2023-05-17:
Going to wait until the end of the week and close this one out

TheTrainGoes, 2023-06-11:
Closing due to non-response and inability to replicate

closed 2023-06-11 by TheTrainGoes