OpenStickCommunity/GP2040-CE #130: ubuntu build failed

chiyu521, 2023-03-30:
pico/GP2040-CE/src/storagemanager.cpp:114:2: sorry, unimplemented: non-trivial designated initializers not supported
pico/GP2040-CE/src/storagemanager.cpp:123:2: sorry, unimplemented: non-trivial designated initializers not supported

chiyu521, 2023-03-30:
[ 84%] Building CXX object CMakeFiles/GP2040-CE.dir/src/storagemanager.cpp.obj
/data/home/chiyu/pico/GP2040-CE/src/storagemanager.cpp: In member function 'void Storage::setDefaultBoardOptions()':
/data/home/chiyu/pico/GP2040-CE/src/storagemanager.cpp:114:2: sorry, unimplemented: non-trivial designated initializers not supported
  };
  ^
/data/home/chiyu/pico/GP2040-CE/src/storagemanager.cpp:123:2: sorry, unimplemented: non-trivial designated initializers not supported
  };
  ^
make[2]: *** [CMakeFiles/GP2040-CE.dir/build.make:160：CMakeFiles/GP2040-CE.dir/src/storagemanager.cpp.obj]

mthiesen, 2023-03-30:
Thiss issue seems to be caused by an outdated compiler. I can build the project without issue on Ubuntu 22.04.2 and arm-none-eabi-g++ 10.3.1.

chiyu521, 2023-03-31:
Thank you, the problem has been solved