OpenStickCommunity/GP2040-CE #147: Web config - Display configuration page updates

FeralAI, 2023-04-15:
The Display Configuration is set up in a very developer-focused manner, with all the technical details explained and available for selection. Several of these options can be simplified, such as:

* The SDA/SCL pins must be used in specific pairings (0/1 on i2c0, 2/3 on i2c1, etc.) and the I2C blocks are also tied to these. All three of these dropdowns can be consolidated into a single dropdown option like `SDA 0 / SCL 1/ I2C0`, etc.
* I2C Address is a text field, should probably be a dropdown with limited selections
* I2C Speed allows arbitrary values with no guidance as to what reasonable values are. Could be a dropdown with limited values or a list of predefined values to populate the text field, which still allows full customization by the user.
* Some options like Flip Display and Invert Display may not be intuitive. A graphic with the preview of the changes here would be nice. It could also integrate with the custom splash display we have.

TheTrainGoes, 2024-02-04:
The implementation of `Peripheral Mapping` has addressed the first and third point.  

Will need to look into the second point more.

Fouth point we can address with some pop ups.

mikepparks, 2024-08-02:
With #1072 the second point (I2C address) is no longer required and deprecated

TheTrainGoes, 2025-01-03:
Closing this - Most has been updated and deprecated

closed 2025-01-03 by TheTrainGoes