OpenStickCommunity/GP2040-CE #162: I made a 3D .step file of the Pico Fighting Board to aid with diy fabrication.

multigamesystem, 2023-04-20:
I've created a to scale 3D model in CAD.  Its in .step format so it's easily loaded into many CAD programs.  The intention is to aid people with designing 3D printed enclosures or CNC cut parts..

[image]

I've uploaded the .step file to my fork of the original Pico Fighting Board repo. 

[https://github.com/multigamesystem/PicoFightingBoard/tree/fd67aa0f9b772f1285bba5c89ccf2695f30f03f0/Step%20file%20for%20CAD](url)

Please excuse my ignorance as I am new to GitHub contribution.

henrebotha, 2023-04-20:
I think this is not appropriate for this repo. It should probably be added (via a "pull request") to the Pico Fighting Board repo.

multigamesystem, 2023-04-20:
> I think this is not appropriate for this repo. It should probably be added (via a "pull request") to the Pico Fighting Board repo.

I've never made a pull request so wasn't sure how to do it.  I did make a similar post on the Pico Fighting Board repo but I think it's no longer being maintained.

henrebotha, 2023-04-20:
I could maybe talk you through it on the OpenStick Discord server. It's not super complicated.

FeralAI, 2023-04-20:
I'll address this in the PicoFightingBoard repo here: https://github.com/FeralAI/PicoFightingBoard/issues/8

This issue can be closed.

closed 2023-04-20 by arntsonl