OpenStickCommunity/GP2040-CE #163: 0.6.3 I found the problem

fangxiang-gan, 2023-04-20:
The firmware I am using is GP2040-CE_ 0.6.3_ PicoFightingBoard.uf2, I found the following problem: I assigned pin16&pin17 to the SOCD Selection Slider and pin18&pin19 to the Joystick Selection Slider. The joystick will experience a random lag. Especially for pin 16, as long as it is used, exceptions will occur, such as assigning it to Dual Directional Input. Are these four sets of pins unusable? I see in the instructions that these four sets of pins are used for player LEDs, but I don't think they are very useful. It would be a good choice to assign them to slider switches.
Thanks.

arntsonl, 2023-04-20:
We will try to replicate on our end, can you send a picture of your setup?

Additionally, can you let us know what kind of sliders you are using for these selections?

Pin 16,17,18, and 19 should be fine for these options.

TheTrainGoes, 2023-04-20:
Hi @fangxiang-gan ,

The issue here is likely that on the Pico Fighting Board there are resistors on pins 16,17,18 and 19 for the intended Player LEDs.  

If possible can you retry this mapping them to some of the normal pins?

fangxiang-gan, 2023-04-20:
[image]
[image]
[image]

fangxiang-gan, 2023-04-20:
The frame I am using is Razer Panthera
[image]

fangxiang-gan, 2023-04-20:
If set as shown in the following picture, the direction will be 100% chaotic. My system is Windows 10 64bit. 
Thank you.
[image]

TheTrainGoes, 2023-04-20:
Hi @fangxiang-gan,

Reposting my previous comment:

The issue here is likely that on the Pico Fighting Board there are resistors on pins 16,17,18 and 19 for the intended Player LEDs.

If possible can you retry this mapping them to some of the normal pins?

---

Can you please also post a pic of the controller board you are using?  

Additionally, there may be issues with how that aux board does its slide switch.  Many factors to consider.

fangxiang-gan, 2023-04-21:
Hi@TheTrainGoes，

Hello, here is the controller chip I am using.
[image]
[image]

I used the following settings and did not find any problem with directional drift, which is normal.
[image]
[image]

thank you.

fangxiang-gan, 2023-04-21:
There will still be problems, as shown above, with double clicking in all directions. -_-

TheTrainGoes, 2023-04-22:
Hi @fangxiang-gan - These 26/27/28 are ADC pins.  Can you please try some of the other pins and see if the behaviour continues?

fangxiang-gan, 2023-04-22:
Hi@TheTrainGoes，
The remaining pins are not allowed to be set to a value of -1. Only these pins 15, 16, 17, 18, 19, 25, 26, 27, and 28 can be used and have all been used. thank you.

TheTrainGoes, 2023-05-10:
Hi @fangxiang-gan ,

Checking in on this - Are you able to replicate this with the new v0.7.1 pre-release?

fangxiang-gan, 2023-05-10:
> @TheTrainGoes 
Yes, 0.7.1. As long as LS and RS are open for 16~19pin, the direction will appear random sticky.
[image]

TheTrainGoes, 2023-05-10:
One last question on this, 

Are you able to replicate this when the unit is not connected to your Razer Panthera?

fangxiang-gan, 2023-05-10:
Yes, as long as LS and RS are on and specified as 15~19, 26, 27, 28pin, the direction will appear random sticky, and turn off completely normal. It has nothing to do with my Razer slider switch, as I tested it directly with the pico chip. 
thank you

TheTrainGoes, 2023-05-15:
Hi @fangxiang-gan - Are you able to replicate this on [v0.7.1-RC2](https://github.com/OpenStickCommunity/GP2040-CE/releases/tag/v0.7.1-RC2)?

fangxiang-gan, 2023-05-16:
Hi@TheTrainGoes 
My test results for 0.7.1-RC2 are as follows: if LS and RS are set to 16pin, 17pin, left analog, or gamepad. The direction is almost collapsed and cannot be used. See the video below. If LS and RS are set to 18pin and 19pin, the direction is close to normal, with an XY rotation flashing every 3 minutes.
Thank you for your continuous follow-up.
https://github.com/OpenStickCommunity/GP2040-CE/assets/128204669/04221685-fc9e-4066-9e1e-d3cdc74c5fe9

TheTrainGoes, 2023-06-11:
Hi @fangxiang-gan ,

Is this issue still present in v0.7.2?

closed 2023-06-18 by TheTrainGoes