OpenStickCommunity/GP2040-CE #168: Bounty - First Input Priority SOCD

TheTrainGoes, 2023-04-22:
Add in First Input Priority (FIP) as an SOCD resolution option in the webconfig. Does not require a shortcut, webconfig option only.

## First Input Priority
(1st Direction + 2nd Direction = Result)
L+R=L
R+L=R
U+D=U
D+U=D

Given Simultaneous Cardinal Directions (Up/Down, Left/Right), you do not get the second directional input until the first is released.

## Example:
So the button sequence 
L, L+R, R, R+L, L

results in the inputs
L, L, R, R, L

ScrubTier, 2023-04-24:
I can take a look at this if no one else is. I've got the code already just need to copy it over, compile and test.

TheTrainGoes, 2023-04-24:
Hey @ScrubTier !

I believe @deeebug already started work on this but I would recommend we all connect and see if it can be worked on together.  

Moving forward I will try to call out who’s taken on what.

deeebug, 2023-04-26:
Sorry, forgot to tag this issue.

closed 2023-04-27 by arntsonl