OpenStickCommunity/GP2040-CE #180: Goofy one...DPAD is suddenly Right Analog Stick?

mrebersv, 2023-04-26:
Not sure what's going on, but as of a few days ago, the dpad on my GP2040 (using pins 0-3 for UDLR respectively) is suddenly mapping to the right analog stick where as previously, it was the dpad.  This has been tested with multiple Gamepad Testers.  I haven't updated the firmware, so undoubtedly, whatever caused this change is related to either an OS/Driver update or me inadvertently doing something that caused it to boot into a different mode.  I've tried booting into DirectInput and Switch modes to no avail.  Instead, left and right are still the right analog stick's L/R, but U/D no longer work in DInput/Switch modes.  Going back to Xinput, nothing is fixed/reset; UDLR is still the right stick somehow.  Any ideas as to what happened to cause something so goofy and how to fix this?

TheTrainGoes, 2023-04-26:
Hey @mrebersv ,

Check out https://gp2040-ce.info/#/gp2040-shortcuts and let us know if S1 + S2 + Down puts you back in DPad mode

mrebersv, 2023-04-26:
That absolutely did it.  Didn't even realize that was a mode that could be selected.  My wife just let me know that my 3yo was playing with my joystick, so he probably just inadvertently did it.  Thanks for the help!

mrebersv, 2023-04-26:
Feel free to close as resolved

closed 2023-04-26 by mrebersv