OpenStickCommunity/GP2040-CE #182: Stick issue

eraldylli, 2023-04-27:
Hello, there.

I've been using this with a stick build. And a weird issue started happening after a couple of weeks, without being provoked by any changes I made. All of a sudden, the stick inputs would not be read. The few first seconds only 1 input every 5 or so would be read, but after a minute it would not read any directional inputs. This happened mid game, but the same problem was detected using the Windows USB Controller tester. Disconnecting USB and reconnecting doesn't fix it most of the times. When it does, it happens again in the next minutes.

But there is something else, if I use directional buttons (stickless way) it works without any problems. I tried using dual Dpad (tried all settings, every combination), even connecting the stick and directional buttons in parallel ('stick up' in parallel with 'button up', 'stick down' with 'button down', and so on...) and the stick commands eventually stop working but the directional buttons (stickless type) keep working perfectly.

I have tried using a different Pico (official), a 16mb one for China, but they problem eventually re-occurs.

I'm including a link to my settings backup file, if its helpful. If not, please disregard. https://we.tl/t-belPSgJF5j

TheTrainGoes, 2023-04-27:
Hello,

This sounds like it might be an issue with your stick.  What kind of joystick are you using on it and do you have the ability to check the inputs from it directly?

TheTrainGoes, 2023-04-28:
Hi @eraldylli ,

This is a strange one.  I have heard of this issue before and it was either a bad wire loom from the joystick or a flakey PCB on the joystick.  

There should be no issues with 3.3v, many people use JLFs on these boards without issue.  

You can try replacing the Pico but I would not expect to see anything change.

eraldylli, 2023-04-28:
I tried another joystick (with its own cable), but couldn't replicate the issue for the 12 hours I left it connected to the pc. However, I think you hit it on the head, its probably bad cable. 
Thank you for your help.

eraldylli, 2023-05-01:
After a few days with no issue, I can confirm it was probably the cables fault.
Feel free to close this issue. And thanks again for your help.

TheTrainGoes, 2023-05-01:
No prob!   Glad to hear it is resolved @eraldylli

closed 2023-05-01 by TheTrainGoes