OpenStickCommunity/GP2040-CE #190: Feature request: Option to make Switch & PS4 layouts align better

henrebotha, 2023-04-30:
The button that PS4 (#185) uses for taking screenshots is not, by default, the same button that Switch uses for the same. I understand there's probably a good reason for this (HID?), but I would like an option to change this behaviour, because the PS4 becomes a bit of a combo breaker, since every other input mode uses S1 as a "select"-like button. 

When this option is enabled, the buttons should align as follows.

| GP2040-CE | Switch | PS4 |
| --- | --- | --- |
| S1 | Minus | Touchpad |
| S2 | Plus | Options |
| A2 | Capture | Share |

In most cases, this will ensure that **the physical layout of the controller stays more constant across input modes**.

This could be set in webconfig only (or even at compile time), but there's an argument to create a runtime shortcut for it too: A user using an adapter device to play on PS4 using a different input mode on GP2040-CE will also have the problem of buttons swapping semantic meanings when compared to Switch, XInput, etc, so in such a case it would also be useful to be able to toggle the behaviour manually.

TheTrainGoes, 2023-05-15:
@henrebotha ,

This is a good argument.  There is nothing preventing this from being a thing.

TheTrainGoes, 2023-05-23:
@henrebotha we will be adding the option in web-config to switch these buttons around in PS4 mode.

TheTrainGoes, 2023-06-11:
This has been added - closing request

closed 2023-06-11 by TheTrainGoes