OpenStickCommunity/GP2040-CE #192: Feature request: Web config on mobile

y3sen, 2023-04-30:
Just wondering if it would be possible to make it so that the web config accessible on a phone. Would love this feature to make it easy to configurate or troubleshoot on the go without the need of a laptop. Thanks

SavageCore, 2023-05-25:
This seems like 2 issues.

Firstly, the Web UI needs to become more responsive, which is no big deal, I started to work on it but will probably finish it after Dark mode.

As for actually plugging into a phone, I'm no hardware dev, never touched TinyUSB but seems [unlikely/luck of the draw](https://github.com/hathach/tinyusb/blob/23c8670e7910ae0987b789c0a640f7cc9ad951f9/examples/device/net_lwip_webserver/src/main.c#L36).

TheTrainGoes, 2023-09-06:
We are going to close this for now as not planned

closed 2023-09-06 by TheTrainGoes