OpenStickCommunity/GP2040-CE #193: Feature request: Independent SOCD cleaning options for X/Y axes

StrangersLove, 2023-04-30:
The 3 SOCD options currently available do not allow for certain SOCD combinations, one important one being Neutral mode on Y axes, and Last Input Priority on X axes. By allowing for more SOCD options, users can customize SOCD options for individual play-style more freely.

If independent SOCD options for X/Y axes is too much trouble/small case usage, please add Neutral mode on Y axes, and Last Input Priority on X axes as a 4th SOCD option as it would help a great many charge character users. Thank you.

TheTrainGoes, 2023-06-11:
Hi @StrangersLove ,

Addition SOCD options is something that pops up from time to time.  For this I would say we would be looking at an `advanced SOCD` mode or similar.  

This would make for a good bounty if you are interested in sponsoring it.

TheTrainGoes, 2023-06-18:
Hi @StrangersLove ,

For the time being we are going to close this as not planned.  If you are interesting in sponsoring a bounty for this please let me know and we will get a new, more detailed submission done.

closed 2023-06-18 by TheTrainGoes