OpenStickCommunity/GP2040-CE #196: Properties being dropped when saving addon settings

mikepparks, 2023-05-01:
After some testing, it seems that the amount of new properties for addons has caused the JSON doc to expand beyond the 2048 limit from LWIP_HTTPD_POST_MAX_PAYLOAD_LEN in webconfig.cpp. This may impact some of the newly added addons, and was only noticed on hardware since the local web server instance does not have this limitation.

mikepparks, 2023-05-01:
[image]
This is a compare against the setAddonsOptions API route. The left is the POST payload. The right is the same doc being returned.

TheTrainGoes, 2023-05-01:
Thanks for the info @mikepparks - We will look into this.

arntsonl, 2023-05-03:
Should be fixed! pull latest and try it out

TheTrainGoes, 2023-05-15:
This has been addressed in [v0.7.1-RC2](https://github.com/OpenStickCommunity/GP2040-CE/releases/tag/v0.7.1-RC2)

closed 2023-05-15 by TheTrainGoes