OpenStickCommunity/GP2040-CE #201: Add instructions for using the Web Configurator on KDE

Khrona98, 2023-05-01:
I was trying to access the Web Configurator on my Linux PC but the page kept returning that it took to long to respond. After testing on my laptop I discovered that KDE sees the controller as a Wired Connection, and automatically connects to it when the controller is pluged in while holding `START`, but when using an already wired connection (like on my main PC), KDE won't automatically connect to the controller embedded web configurator.

The solution is rather simple and takes only around a minute, but someone that stumbles on this same problem might not find why it doesn't work automatically. So, basically:

- Connect the controller to your PC while holding `START`;
- Open `System Settings` > `Connections` | Alternatively, click on the `Networks applet` (Computer with a cable icon) on the `System Tray` > click on the `Configuration Button` on the top-right;
- Click the `+` button;
- Click on `Wired Ethernet` and then on the `Create` button at the bottom of the window;
- Click on the down arrow in `Restrict to device`;
- Choose the one that says `enp11s0f3u1` (or something like that, yours could be different), don't choose `eno1`, that's your primary connection;
- Click on the `Save` button;
- Done.

(Optional) On the `Wired` tab of your main connection (`Wired Connection` in the list on the left), choose eno1 in the dropdown menu for `Restrict to device` and click `Apply`. This way it won't show duplicated entries when you open the `Networks applet` on the `System Tray` while having the controller connected in Web Configurator mode.

InfraredAces, 2023-05-02:
I run Arch with KDE Plasma (would need to look at specific versions) and I have not experienced this. It is able to be connected to my network via Ethernet and still connect to the WebConfig off of the Pico with no further alterations. They just appear as 2 separate wired connections

Khrona98, 2023-05-02:
> I run Arch with KDE Plasma (would need to look at specific versions) and I have not experienced this. It is able to be connected to my network via Ethernet and still connect to the WebConfig off of the Pico with no further alterations. They just appear as 2 separate wired connections

I think that for me doesn't connect automatically since it doesn't detect the Pico as a new standalone connection. I can see the device when I open the applet, but as a duplicate of my main connection and can only have one or the other active at a time, unless I do what I described in the issue.

For reference I'm on Plasma `5.27.4`. It could very well have different behavior on another version.

davewongillies, 2023-05-02:
I'm a Linux user who this particular problem doesn't effect either (I don't use KDE Plasma but it shouldn't make a difference since this a networking/network-manager issue). It has also come up in the past in discord and there were some people who this effected and others who it didn't.

There's likely to be some routing weirdness going on with the people who it effected, possible that their local subnet also uses 192.168.7.0/24?

davewongillies, 2023-05-02:
> I think that for me it doesn't connect automatically because it doesn't detect the Pico as a new standalone connection. I can see the device when I open the applet, but as a duplicate of my main connection and can only have one or the other active at a time, unless I do what I described in the issue.

@Khrona98 Could run the command `ip route` from a terminal and post the output in this issue? Run it twice, once with your regular connection active and again with the web configurator connection active.

Khrona98, 2023-05-02:
> @Khrona98 Could you run the command `ip route` from a terminal and post the output in this issue? Run it twice, once with your regular connection active and again with the web configurator connection active.

Pico disconnected:
```
default via 192.168.1.254 dev eno1 proto dhcp src 192.168.1.78 metric 100 
192.168.1.0/24 dev eno1 proto kernel scope link src 192.168.1.78 metric 100 
```

Pico connected (with the configuration I cited in the original post):
```
default via 192.168.1.254 dev eno1 proto dhcp src 192.168.1.78 metric 100 
192.168.1.0/24 dev eno1 proto kernel scope link src 192.168.1.78 metric 100
192.168.7.0/24 dev enp11s0f3u1 proto kernel scope link src 192.168.7.2 metric 101  
```

I did find something else too. If I delete the config I did for the Pico in the settings and connect, it shows exactly the same as the first screenshot, but if I Restrict my main connection to en01 and then connect the Pico, it works correctly showing as another connection and connects automatically every time without the workaround I described. It also shows this output:

```
default via 192.168.1.254 dev eno1 proto dhcp src 192.168.1.78 metric 100 
192.168.1.0/24 dev eno1 proto kernel scope link src 192.168.1.78 metric 100 
192.168.7.0/24 dev enp11s0f3u1 proto kernel scope link src 192.168.7.2 metric 101
```

TheTrainGoes, 2023-09-06:
This has been added into the documentation - closing

closed 2023-09-06 by TheTrainGoes