OpenStickCommunity/GP2040-CE #202: 0.7 Turbo problem

xboxgod, 2023-05-02:
new 0.7 version default Turbo pin is -1 , I try to use up or down arrow button restore to 0.62 ver Pin in 14, 
up arrow button give me red prompt : 0 is already assigned!
but older version u can continue choose to a empty pin, and this 0.7 ver I can not use it. (so this is bug 1)

next I use keyboard input 14 (after this u can use up or down arrow button choose number, but if u let it to -1 bug 1 will show again)

The most important question is now I am setting to:

Turbo Pin: 14
Turbo Pin LED: -1
Turbo Shot Count: 15
Turbo Dial (ADC ONLY): -1

and let Turbo Add On Enable , when I reinsert usb cable, OLED displayer can show T15
but I press Turbo Button (in Pin 14)

None of the buttons that could be turbo in the previous version did not work (This is bug 2)

I did someting wrong or is there really a BUG?

fangxiang-gan, 2023-05-02:
This is a bug.The turbo cannot be started. PicoFightingBoard.uf2 It's the same.After the web configuration is enabled, it will still show the closed state when you enter it again.

TheTrainGoes, 2023-05-02:
Yes we are aware of the bug and will work to address it.

closed 2023-05-15 by TheTrainGoes