OpenStickCommunity/GP2040-CE #224: Add a way to backup PS4 keys

sgfdevices, 2023-05-07:
If I've loaded a PS4 key on my device, but need to reflash firmware, it would be ideal to be able to backup the keys on the device. Perhaps included with a regular backup from webconfig? Or some other way.

arntsonl, 2023-05-07:
We're currently looking at switching our save/upgrade system to a new protobuf base, which should preserve settings between firmware upgrades. I'll keep this open for now

TheTrainGoes, 2023-05-15:
Hey @sgfdevices ,

I am going to split this into two separate asks:

#1 - The ability to preserve keys through a normal firmware update
#2 - The ability to backup keys which can be reloaded

For #1, we are moving to a place where this can happen.  The hope is that at some point soon updating the firmware will be simple drag and drop that will not require re-uploading anything or re-doing custom settings.  

For #2, we do not feel it is the right course of action to create something that can be backed up and distributed.  

We will close this request as `won't do` based on the backup request, but know that we are working to have settings retain through firmware updated.  

* note that this does not apply for firmware updates which require a nuke first or any other instances where the nuke must be applied to recover a pico.

closed 2023-05-15 by TheTrainGoes