OpenStickCommunity/GP2040-CE #229: Dual Direction not working properly

ez2torta, 2023-05-08:
How to replicate:

Set 4 pins for dual input thing.
Test on gamepadtester and/or others pressing both your joystick and dual input

[image]

What i expected:

- Those pins when shorted to ground should move the left stick

What i got:

- They just duplicate whatever was on the joystick

Tested this using Left Analog stick as Dpad Mode in config too, and the results were the same (they clone whatever was in the dpad pins)
[image]

ez2torta, 2023-05-08:
just edited the main post to make things a little bit more clear

bsstephan, 2023-05-08:
Ah, okay. So Combination Mode: Mixed combines the two sets of directionals into whichever mode D-Pad Mode is set to. It sounds like you're closer to getting what you want if you are in Combination Mode: None, which has the extra set always register as Dual D-Pad Mode, so you could have the "normal" inputs set to left stick and the "dual" inputs set to dpad, and they register separately. BUT,

Caveat 1: None mode doesn't work on 0.7.0. The fix I wrote is merged as of yesterday, but a new release hasn't happened yet. So you want to wait for the next release or build your own firmware from main.

Caveat 2: there's no way to change Dual D-Pad Mode on the fly, so if you're saying you want to toggle both arbitrarily, you can't do that (there's no software macro or wiring for two toggles). But if you're happy to use the web configurator to change the dual mode, or are happy with one of your inputs always been in a certain mode, then I think Combination Mode: None is what you want.

bsstephan, 2023-05-08:
I wrote a table in the PR for my Combination Mode: None fix here: https://github.com/OpenStickCommunity/GP2040-CE/pull/220 --- you can reference it to see what it accomplishes and see if that lines up with what you want.

ez2torta, 2023-05-08:
it seems that your PR covers both scenarios i wanted :) thanks!

ez2torta, 2023-05-08:
Just compiled and it's working properly as you said. Thanks! i hope everyone can hop into the new version soon

closed 2023-05-08 by ez2torta