OpenStickCommunity/GP2040-CE #231: Webconfig should indicate if keys have been loaded

sgfdevices, 2023-05-09:
If you load keys onto a device in the addons section, then reboot and go back to webconfig, there's no indication that keys have been loaded, just the ability to load new ones, like there aren't any to begin with. It would be ideal to have a way to confirm that keys exist before potentially overwriting them.

TheTrainGoes, 2023-05-10:
Hey @sgfdevices  - Good callout.  We will look into this.

aserrallerios, 2023-05-17:
I'm closing https://github.com/OpenStickCommunity/GP2040-CE/issues/245 as it seems to be dupe of this issue right here.

arntsonl, 2023-06-12:
We could add an end-point that validates of PS4 keys are set to null or not. Or even /api/ps4AuthEnabled true/false?

That would probably do it, I might add this.

TheTrainGoes, 2024-03-28:
We are going to close this as not-planned.  

We believe most people are using passthrough now and in order to have this shown in web-config we would need access to the keys which introduces complexities we are not comfortable with.

closed 2024-03-28 by TheTrainGoes