OpenStickCommunity/GP2040-CE #240: Dual Directional input not working as intended

KurtMage, 2023-05-11:
I want to have one set of buttons correspond to the dpad and another set be the left analog stick, similar to the [Cross Up](https://www.hitboxarcade.com/products/cross-up). Based on documentation, it looks like [dual directional input](https://gp2040-ce.info/#/web-configurator?id=dual-directional-input) should get me what I want.

In `Settings` I set the regular buttons to Dpad and in the `Dual Directional Input add-on`, I set it to left analog stick. Behavior was different in different modes, but none were what is described in documentation. What I observed was:
`Mixed`: Both sets of buttons give Dpad output.
`Gamepad`: the dual directional input add-on buttons give left analog stick output (yay), but the regular buttons give both Dpad and left analog stick output simultaneously.
`Dual directional`: I actually do not remember what I saw here, but it was definitely not what I wanted.
`None`: The dual directional input add-on buttons are disabled entirely.

bsstephan, 2023-05-11:
Hello, I recently fixed DDI mode `None` in PR #220 --- please take a look at my notes there to see that it accomplishes what you want, but your use case sounds the same as mine. It is in the mainline code and will be in the next release, AFAIK.

KurtMage, 2023-05-11:
That's great to hear! I'll keep an eye out for the new release. Thanks!

TheTrainGoes, 2023-05-11:
Sorry @KurtMage - had to edit your original post to change a name in there.

bsstephan, 2023-05-18:
@KurtMage - want to try v0.7.1 now?

TheTrainGoes, 2023-05-23:
Closing due to lack of reply - should be fixed in PR#220

closed 2023-05-23 by TheTrainGoes

KurtMage, 2023-05-24:
Finally got around to testing. Looks good to me! Thanks!

TheTrainGoes, 2023-05-24:
Closing - Issues resolved in v0.7.1