OpenStickCommunity/GP2040-CE #242: Issue with up/down on Ps3, and asking for compatibility on Ps2

Pabletex91, 2023-05-12:
I just replaced my old "zero lag" board with a original pi pico, works waaay better on my pc, but in my ps3 (I realized this on the xmb, but no in games) when I move up or down, sometimes it gets stuck until I move again.

I also want to know if you could add Ps2 compatibility through OPL and Pademu, I plugged in Ps3/Ps4 mode and no response, my Ps2 would be very happy :)

Thanks in advance

henrebotha, 2023-05-12:
Best to split up issues. It helps developers keep better track of progress. I suggest you edit this issue & title to just cover the up/down bug, and create a new issue for the PS2 compat.

wrennnnnn, 2023-05-22:
the first issue happens on original hardware (dualshock 3 and original ps3). not sure why it happens, though.

TheTrainGoes, 2023-05-23:
@Pabletex91 based on the comment from @wrennnnnn the first issue is normal behaviour.  

For the 2nd comment about PS2 support please feel free to file a new issue and possibly sponsor a bounty around it.

closed 2023-05-23 by TheTrainGoes