OpenStickCommunity/GP2040-CE #243: Dark-Mode for Web-Config

lost-in-spacetime, 2023-05-12:
Playing around last night in the web-config made me realize how bright the page can be when full screen at night.

I'm currently using the plugin Dark Reader to remedy the issue for myself. 

It was suggested that I create an issue, possibly an easy update to add Dark-Mode to the config? 

Thanks for entertaining my idea!

TheTrainGoes, 2023-05-15:
This sounds like a good bounty candidate.  Dark mode slider?

FeralAI, 2023-05-28:
Very nice @SavageCore! I'm ok bringing in the Bootstrap v5.3 alpha dependecy for now. Our use of Boostrap is quite limited, so I doubt rolling with this version would be an issue until the stable version is released.

Just a few things and I think this could be a PR:

* Could `DarkMode.js` be renamed to `ColorScheme.js`? That seems more fitting since this could in theory be used for a variety of themes.
* We've started to consolidate localStorage options in `AppContext.js` and the `theme` setting would go there

closed 2023-06-18 by SavageCore