OpenStickCommunity/GP2040-CE #247: Bounty - Bluetooth support using the Pico W

Fortinbra, 2023-05-13:
We would like to have support for Bluetooth and the Pico W board.

### Requirements
- [ ] **MUST** maintain USB compatibility
- [ ] **MUST** not increase USB latency
- [ ] **MUST** support at least Switch and X-Input modes via Bluetooth.

### Addition Considerations
Not required, but would be nice to have.
- Pimoroni Lipo Amigo Pro battery management.
   https://shop.pimoroni.com/products/lipo-amigo?variant=39779302539347
- Web Config over Bluetooth.

TheTrainGoes, 2023-05-15:
We chatted about this and are going to break it down into two separate bounties.  

Part 1 will be a proof of concept.  This proof of concept is an MVP to see if it is even viable as a controller and has reasonable lag / performance / etc.  

This part can be quick and easy.

Part 2 will be the full incorporation into GP2040-CE.

Fortinbra, 2023-06-15:
> Part 1 should be covered by PR [bluekitchen/btstack#209](https://github.com/bluekitchen/btstack/pull/209).
> 
> Extending the HID for more buttons,axis and modes could be seen in https://github.com/lemmingDev/ESP32-BLE-Gamepad
> 
> and possibly ForceFeedback with https://github.com/YukMingLaw/ArduinoJoystickWithFFBLibrary

While this is great information to have. What I'm looking for, for the bounty, is someone to actually make a demonstration using our codebase as the example.

dogtopus, 2023-06-20:
I didn't know Pico W also supports Bluetooth Classic (a crucial thing to have for emulating most console Bluetooth controllers) until now.

We might actually get somewhere with this bounty. GP2040 DIY handheld controller anyone?

dogtopus, 2023-06-22:
> Since the chip is connected via SPI... could this eventually be ported over to use ESP?

In theory yes but we will eventually run into the issues with Bluekitchen's license agreement since it's not technically free software and from what I heard from them years ago, requires commercial licenses even if the firmware is open source but loaded officially to a board (open source or not) for sale.

hizzlekizzle, 2023-06-22:
Bluetooth support would also open the door for DIY receivers to connect bluetooth gamepads to S/NES or Wii extensions, right? That is, using the board as a receiver rather than a transmitter; similar to the SNES 'retro receiver' from 8bitdo?

Fortinbra, 2023-06-22:
> Since the chip is connected via SPI... could this eventually be ported over to use ESP?

For GP2040-CE, we don't have any plans to support any platforms other than the RP2040. You're always welcome to fork, and do your own thing.

Fortinbra, 2023-06-22:
> Bluetooth support would also open the door for DIY receivers to connect bluetooth gamepads to S/NES or Wii extensions, right? That is, using the board as a receiver rather than a transmitter; similar to the SNES 'retro receiver' from 8bitdo?

We need to get transmission working first.

dogtopus, 2023-06-22:
> > Since the chip is connected via SPI... could this eventually be ported over to use ESP?
> 
> For GP2040-CE, we don't have any plans to support any platforms other than the RP2040. You're always welcome to fork, and do your own thing.

I think they meant using ESP as an HCI, which I'm not sure if the license for the Bluekitchen stack in pico-sdk permits such use case.

Fortinbra, 2023-06-22:
> I think they meant using ESP as an HCI, which I'm not sure if the license for the Bluekitchen stack in pico-sdk permits such use case.

Oh, in that case, we're only utilizing what is available from the SDK, and we'll deviate after we have something working, if we have reason to.

futaris, 2023-06-26:
I know I said I'd have this working last month.  I ended up ordering a couple more Pico Ws so I can implement this.

sanjay900, 2023-06-30:
> Bluetooth support would also open the door for DIY receivers to connect bluetooth gamepads to S/NES or Wii extensions, right? That is, using the board as a receiver rather than a transmitter; similar to the SNES 'retro receiver' from 8bitdo?

Just as a sidenote, having done this for a project of my own, there is a bug with the current Bluetooth driver that limits pico to pico hid polling pretty badly, though i haven't had time to look at alternatives to that. https://github.com/raspberrypi/pico-sdk/issues/1342

futaris, 2023-07-16:
https://github.com/DavidPagels/retro-pico-switch

aallan, 2023-07-18:
> (Also take this with a grain of salt since looks like https://github.com/raspberrypi/pico-sdk/issues/1164#issuecomment-1372677903 was to make it available for use with RP2040 and not just Pico W. It's better to formally ask them once we actually need to use the stack on non-Pico W boards.)

While the CWY43439 driver is [licensed for commercial use](https://github.com/georgerobotics/cyw43-driver/blob/195dfcc10bb6f379e3dea45147590db2203d3c7b/LICENSE.RP) for users of Pico W, or anyone else who builds their product around RP2040 and CYW43439, this is not the case for the BlueKitchen license.

In addition to the [standard BTstack licensing](https://github.com/bluekitchen/btstack/blob/master/LICENSE) terms, a [supplemental license](https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2_common/pico_btstack/LICENSE.RP) which covers commercial use of BTstack with Raspberry Pi Pico W or Raspberry Pi Pico WH has been provided. There is no exception for those using RP2040 on their own boards.

dogtopus, 2023-07-18:
@aallan Thanks for the clarification. I'm personally not a fan of this licensing model but I guess that's to be expected when using non-free libraries like bluekitchen stack.

TheTrainGoes, 2023-09-06:
This issue is going to be re-written to better reflect the goals of the project.  I will keep it open until the new issue is posted.

jars28, 2024-02-21:
https://github.com/DavidPagels/retro-pico-switch

Rough implementation of BlueTooth support with Switch connection is in this project.

henrebotha, 2024-02-21:
> Rough implementation of BlueTooth support with Switch connection is in this project.

Unfortunately, it uses the Pico SDK, which I believe we can't use.

Invictaz, 2024-02-21:
> Bluetooth support would also open the door for DIY receivers to connect bluetooth gamepads to S/NES or Wii extensions, right? That is, using the board as a receiver rather than a transmitter; similar to the SNES 'retro receiver' from 8bitdo?

We already have that on ESP32 in Blueretro but there must be a reason that Blueretro doesn't get ported to Pico. Maybe because of limited flash storage or the lack of speed.

dogtopus, 2024-02-21:
> We already have that on ESP32 in Blueretro but there must be a reason that Blueretro doesn't get ported to Pico. Maybe because of limited flash storage or the lack of speed.

It's the same reason why GP2040 is not ported to ESP32. Because it's heavily BSP-specific and effectively not portable without major time and effort spent to rewrite the code.

jars28, 2024-02-21:
"Raspberry Pi Pico W customers benefit from a pre-paid commercial license for BTstack, allowing you to use Pico W as a component in your own products."

https://www.raspberrypi.com/news/new-functionality-bluetooth-for-pico-w/

Fortinbra, 2024-02-21:
> > Rough implementation of BlueTooth support with Switch connection is in this project.
> 
> Unfortunately, it uses the Pico SDK, which I believe we can't use.

The Pico Sdk is exactly what we're using, it's why we can't use other drivers for Bluetooth at the moment, only BlueStacks, since it's already part of the SDK.

bsstephan, 2024-02-21:
Just adding citations here:

The driver is normally a non-commercial piece of software that has been licensed to RPi by BlueKitchen, and RPi has in turn licensed its free use only if you are creating a Pico-W or Pico-W derived product.

* BTstack license: https://github.com/bluekitchen/btstack/blob/master/LICENSE
* the pico-sdk exception: https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2_common/pico_btstack/LICENSE.RP

Both licenses lead to a scenario where the usage of BTstack would arguably (if not clearly) limit GP2040-CE to non-commercial applications and/or Pico W-only applications, so it's not an option for us at this time.

sanjay900, 2024-02-21:
By the way, i did ask about this for one of my projects and got this response back:

At the moment, Raspberry Pi has license our stack for all of their Pico W boards. By this, the Pico W can be used in commercial products without a license from us.
If a RP2040 with some other Bluetooth Controller is used, this requires to sign a license agreement with us.

On the other hand, if you build your own board with a RP 2040, you still need a Bluetooth Controller to use Bluetooth. The Pico W uses a CYW43439 and modules with the are more expensive than a Pico W. If you order 1k of CYW43439 from Mouser/Digikey, you still pay 3.80 USD. If you buy RP2040 on a reel, that's 0.63 USD. This gets you already to about 4.50 USD and you still need a flash, oscillator, ... so, unless there's a good reason, it's easier and cheaper to just buy Pico Ws :)

If you want to be cheaper than the Pico W, I would suggest the ESP32 (BR/EDR/LE/Wifi) or the ESP32-C3 (LE/WIFI). You can either use the Bluedroid or NimBLE stack that is free to use, or ours with a license.
When you get closer to choosing a path, feel free to get back and discuss your licensing needs. We try to make our stack affordable to small startups/Makers with low volumes.

TheTrainGoes, 2024-02-21:
This is an odd line worth more investigation:

"If a RP2040 with some other Bluetooth Controller is used, this requires to sign a license agreement with us."

Mind reaching back out to see if the license is handled should we choose to use the same parts as the W?  

Why would I do that if it costs more than the Pico W?  Easy!  I don't want to solder a bunch of them and I need more GPIO.

sanjay900, 2024-02-21:
You would 100% have to license it with them if you use anything that isn't a PicoW, even if the parts are the same.

TheTrainGoes, 2025-01-03:
We are going to close this issue and make a new one if we are still interested in looking at Pico W support.

closed 2025-01-03 by TheTrainGoes

Invictaz, 2025-01-03:
We have Bluepad32 that is also on OGX Mini which is sold commercially?

TheTrainGoes, 2025-01-03:
Sorry @Invictaz I do not understand your comment.  

We are looking for something that can used on either the Pico W or via embedded components.  The issue is with embedded components.  

We also have not seen much interest in BT support from the community.

Invictaz, 2025-01-04:
Maybe then do a poll then you will know.
Bluepad32 works perfectly on Pico W.
See ogx-mini on Github

Fortinbra, 2025-01-04:
Bluepad32 is a host framework, we need to be a client.