OpenStickCommunity/GP2040-CE #256: Save Button is Not Working Pico 0.7.1 On Add-On Screen

mytokat84, 2023-05-16:
I m trying to save turbo button pin in add on screen and save button is not working. I tried it on Chrome and Microsoft Edge.

FeralAI, 2023-05-17:
Hello @mytokat84,

Sorry you're having issues saving on the addons page. We've identified a possible issue related to the Analog addon causing an issue on save. Do you have that addon enabled? If not we'll need to look into this a bit further.

mytokat84, 2023-05-18:
Hello FeralAI, 

First of all, I would like to thank you on behalf of thousands of people for giving us freedom in both the Pico Fighting Board and the GP2040 software.

Although the Analog Add-On option is disabled for me, the save button does not work.

TheTrainGoes, 2023-06-11:
This should now be resolved.  

@mytokat84 if you are able to replicate on the new v0.7.2 RC please let us know and open a new ticket.

closed 2023-06-11 by TheTrainGoes