OpenStickCommunity/GP2040-CE #257: Add-Ons with Input Modifiers cannot currently effect boot mode

arntsonl, 2023-05-17:
@RobertDaleSmith brought up an interesting point:

add-ons that modify inputs such as GPIO expanders cannot currently change the boot mode.

This is because add-ons are loaded after the boot-mode is checked.

Either move add-on loads to before boot mode, OR have add-ons with early load and add-ons with normal load.

arntsonl, 2023-08-24:
Fixed with @NickGuyver's code!

closed 2023-08-24 by arntsonl

RobertDaleSmith, 2023-08-29:
I tested the latest update with my SNES2USB adapter and the boot options work great now. But it seems the other hotkeys are not. For example changing the d-pad to either left or right analog is still not working with add-ons. Did you run into this as well @NickGuyver ?

arntsonl, 2023-08-29:
Hmmm sounds like another issue maybe? Could you open a new issue and we'll figure it out :)