OpenStickCommunity/GP2040-CE #258: read accelerometer/gyro and output value to joystick axis.

MechatronikMonkey, 2023-05-17:
To detect nudging, position, rotation etc.
Some people use this firmware for Virtual Pinball Tables where physical inputs have great impact on the game.

Most ACC or gyro Chips out there for Arduino come with pretty good drivers in C/C++. Should be easy to implement.

I am Embedded C/C++ Dev.... if someone gives me a little help on the software structure and architecture, I can code this function.

henrebotha, 2023-05-19:
This would be an add-on, probably. Here are some examples of existing add-ons that use analogue output.
- https://github.com/OpenStickCommunity/GP2040-CE/blob/main/src/addons/analog.cpp
- https://github.com/OpenStickCommunity/GP2040-CE/blob/main/src/addons/i2canalog1219.cpp
- https://github.com/OpenStickCommunity/GP2040-CE/blob/main/src/addons/wiiext.cpp

TheTrainGoes, 2023-09-06:
We are going to close this request due to lack of interaction and revisit when we have the expansion board finished.

closed 2023-09-06 by TheTrainGoes