OpenStickCommunity/GP2040-CE #260: Shortcuts using Extra Button do not register

InfraredAces, 2023-05-18:
### Prerequisites

- [ X ] I am running the latest version
- [ X ] I checked the documentation and found no answer
- [ X ] I checked to make sure that this issue has not already been filed
- [ X ] I'm reporting the issue to the correct repository

### Expected Behavior

When inputting a shortcut using an input assigned with the Extra Button Add On, the appropriate shortcut should execute, matching the behavior of the original up input.

### Current Behavior

When inputting a shortcut using an input assigned with the Extra Button Add On, the appropriate shortcut does not execute, not matching the behavior of the original input.

### Steps to Reproduce

1. Enable the Extra Button Add On
2. Assign a button's GPIO pin to Extra Button Pin
3. Assign an input to the Extra Button
4. Press shortcut using assigned Extra Button
5. Shortcut fails to register

### Context

- Firmware Version: 0.7.1
- Operating System: Arch Linux
- Browser: Chrome

deeebug, 2023-05-18:
Thanks for reporting this issue. Addressing #259 also here, it's how these addons are reading the pins directly and setting the state. It seems like we need a layer of abstraction to handle this, which I want to take a look at.

You might also encounter this issue in other addons like BOOTSEL button addon.

TheTrainGoes, 2024-02-04:
@InfraredAces this is no longer an issue with the new pin mapping system.  Closing.

closed 2024-02-04 by TheTrainGoes