OpenStickCommunity/GP2040-CE #268: Alternate functions for the Turbo pin/button. Namely a 'menu lock'.

fatihG, 2023-05-21:
I'd like to preface this with that I have not used the software/web tools at all yet, so I am not actually sure if this works or not. 

That being said, I was thinking whether it would be possible to set an alternate function via the webtool/firmware for the turbo buttons. 

I have asked for feedback, on something I am working on, from a few people and one of the things I heard is that they are worried about hitting the menu buttons on accident. From what I have seen we can set the buttons in the webtools.
 
If one of the functionality was to 'lock' the menu buttons, could that be set to the turbo button? A simple toggle. 
When you hit the 'lock' button, all of the menu buttons do not function anymore. (Home, share, menu, touch, L3 and R3.)
When you press it again the lock gets lifted and you can use them again. 

Would it be possible to set certain button combinations to the 'turbo' button/pin? 
Like booting into the web ui, instead of needing to press a button combination you just press a single button that does the same thing?

TheTrainGoes, 2023-05-23:
Hi @fatihG ,

We have talked about this request a few times in the past.  At the time we were more focused on core features but I think we can take another spin at this.  

In a basic implementation this will need either a latching switch or a slide switch that has two positions.  

I will host a bounty for this.

TheTrainGoes, 2023-05-23:
Bounty:

Aux button lockout add-on will allow a slide or latching switch connected to a GPIO pin to dictate other buttons which will no longer respond to presses.  

As an example, when the Aux button lockout add-on is enabled and the slide switch is in the on position, S1, S2, A1, A2 no longer respond to presses.  

The web-config section for this add-on should allow you to set the following:
- Aux button lockout add-on pin
- Any number of buttons that can be affected by this add-on

I am open for discussion on if this should prevent someone from being able to enter web-config should they have S2 locked out.

TheTrainGoes, 2023-06-12:
This work has been completed and the issue will be closed.

PR #275 (2023-06-19): Implement Focus Mode addon
Resolves #268.

This is a new addon that lets given inputs to be negated while the specified pin is held. Currently, arbitrary buttons can be set via the web-config.
closed 2023-06-12 by TheTrainGoes