OpenStickCommunity/GP2040-CE #269: R2 And L2 Buttons Causing Some Problems

mytokat84, 2023-05-22:
Hello, 

In 0.7.1 Pico software, when I press the R2 or L2 button in Ps4 mode, other keys are also pressed.

bsstephan, 2023-05-22:
Can you call out which other keys? What problems are caused? Note that pressing L2/R2 now activates the analog axes assigned to those buttons, but that is normal behavior generally speaking.

mytokat84, 2023-05-22:
> Can you call out which other keys? What problems are caused? Note that pressing L2/R2 now activates the analog axes assigned to those buttons, but that is normal behavior generally speaking.

Yep analog axes. Sorry if this is normal. Thanks for quick reply.

bsstephan, 2023-05-22:
No worries! XInput mode works the same way; it was added to PS4 mode in 0.7.1.

mytokat84, 2023-05-22:
Thanks, 

In Xinput mode I can use R2 and L2 in fighting games perfectly. In ps4 mode and Mame I cant.

TheTrainGoes, 2023-05-22:
@mytokat84 Are you on the latest version of the firmware?  There was a small bug a while back that would cause this to happen.  Can you please check the add-on section of the web-config and see if there are any other add-ons that are turned on?

mytokat84, 2023-05-22:
> @mytokat84 Are you on the latest version of the firmware? There was a small bug a while back that would cause this to happen. Can you please check the add-on section of the web-config and see if there are any other add-ons that are turned on?

Hello @TheTrainGoes. Yep, I m on latest firmware. I tried this with original dual sense and original DS4 and all of them are same. So this is not a problem for me from now on. Thanks for the reply.

closed 2023-05-22 by mytokat84