OpenStickCommunity/GP2040-CE #274: Multiple profiles, switchable using button combinations.

fatihG, 2023-05-23:
### Have you checked the GP2040-CE documentation [HERE](https://gp2040-ce.info/#/) to confirm this feature doesn't already exist?
Yes.

### Which problem is this feature request solving?
A way to quickly change between certain button layouts, without needing to access the web UI. 

### Describe the solution you would like to see
For example when using a set up for fighting games, have a profile with UP = jump. 
But then when playing a platformer has a profile with an action button like LK = jump.

Or another one is a set up tied to specific fighting games. Which could be tied in to the profiles of the existing LED layouts of the various fighting games. 
Have a set up for SF, Tekken, GG etc.

-
While it is not crucial to this functionality, a nice to have would be to have a LED layout tied to these profiles as well. 
So you can easily tell by the LED layout on which profile you are or what game it is set to. 
This could be a simple 'flash' and it goes back to the LED layout that was already set. 
Along with that if you keep holding the button combination for 3 seconds or something, it would make the accompanied LED layout stick. 

If something like this is implemented, it would be nice to have a toggle in the web UI for the profiles, to swap the functionality of the 'flash'  and the LED layout sticking. 

### Describe alternatives you've considered
From what I understand it is possible to do something similar using backups, but that would require needing access to a PC and booting into the Web UI. 

### Can you submit a pull request?
Don't think so? 

### Are you planning on working on this feature yourself?
No. I'd be willing to help out with testing or design.

TheTrainGoes, 2023-05-24:
Hi @fatihG ,

Would you like to sponsor this as a bounty?

TheTrainGoes, 2023-05-25:
I am going to break this down into two separate requests.  

Part one will be a button that can be pressed to change through the profiles.

Part two will be adding LED functionality.

bsstephan, 2023-06-16:
I have a plan for part one. If no one minds I'm going to assign this to myself, so that if anyone else is looking at it, they know to ping me.

TheTrainGoes, 2023-06-21:
This has been a hot topic in the Discord.   

To set the right expectations we would like to keep the initial implementation of this limited to just the 8 main input buttons and nothing else.  

That would be:
A, B, X, Y, L1, L2, R1, R2

Once you get into profile switching of Aux buttons things get tricky.

bsstephan, 2023-06-28:
> This has been a hot topic in the Discord.
>
> To set the right expectations we would like to keep the initial implementation of this limited to just the 8 main input buttons and nothing else.
>
> That [...]

Ope, I just saw this comment.

Current implementation allows all the pin mappings, but I am rejiggering my implementation anyway and can try to encapsulate the above limitation, because I agree, it's *hellishly* confusing to operate the stick properly when you end up changing Sx and Ax buttons. Especially since you probably use them in the hotkeys and then you need to remember what profile you're in to change profiles... it's a lot.

General update: switching works very well as a hotkey action. I need to redo config structure for the above and to hopefully make the React UI easier for my lizard brain to enhance. In general: feature's coming, I just need a bit of time. =)

bsstephan, 2023-07-05:
@fatihG - if you are able to, please check out my branch in https://github.com/OpenStickCommunity/GP2040-CE/pull/384 to see if it meets the expectations and/or there's anything missing that you'd like to see.

bsstephan, 2023-07-06:
After a discussion with @TheTrainGoes I'm going to add the core directions to what can be remapped in a profile. Still not touching the other buttons with a ten foot pole. =)

bsstephan, 2023-08-07:
@fatihG I forgot to ping you, but this has been merged to main, please test it out and let me know what you think, where documentation is lacking, etc.

TheTrainGoes, 2023-09-05:
This work has been completed

closed 2023-09-05 by TheTrainGoes