OpenStickCommunity/GP2040-CE #276: Retro console output adapter/intermediary board

bsstephan, 2023-05-23:
I would like to utilize the GP2040-CE feature set on non-USB hosts, i.e. consoles with serial-based connections such as NES, SNES, PS1/PS2, or Dreamcast. Other boards out there use an RJ45 connection with cables with specific pinouts for each host device (including USB); this implementation would be ideal (especially if it used the same cable pinouts) but is not required.

This should be the only connection point necessary to the host, so I think this device would have to sit in front of the USB connection on a RP2040-based board ("the primary board"), and either pass USB through both directions blindly, or for the other consoles, power the primary board and retrieve the gamepad state from the primary board and reinterpret it for the host console at a polling rate appropriate for the host console. I think this means that core GP2040-CE functionality (SOCD, DDI, etc.) lives on the primary board, and this solution yields the creation of an intermediary board that has its own RP2040 and GP2040-CE variant to implement the classic console timing and overall serial connection, and handle other miscellany like button remaps, re-SOCDing multiple directions, etc. for the host.

I believe this makes more sense as a separate board with a GP2040-CE variant (or fork?) than trying to winch this all into the one firmware and use up its GPIO, burden the primary board build with a bunch of weird timings, etc.

Most interesting to me, and probably a long way towards a solution, is starting work on an implementation for the SNES and Analogue Super Nt. I would consider that console this feature's primary focus, and the work should be able to be expanded upon for other consoles.

I think the core deliverable should be an open design of the new hardware, and whatever firmware changes are necessary to implement the communication between devices and subsequent SNES implementation. Ideally, GP2040-CE enhancements don't affect the main build but are instead contained to a different build, but maybe it'd be easier to call just this a GP2040-CE RETRO fork or something.

I hope to assist on the firmware side of this as time allows, but I'm throwing this out there to hope to get the base hardware realized without me, because I don't know much about designing a PCB and etc. anyway.

TheTrainGoes, 2023-05-23:
I am converting this to a bounty and will sponsor it.

TheTrainGoes, 2023-09-06:
We are going to re-do this issue and break it down by specific console while giving some overall guidance on how we would like to see each system added.

TheTrainGoes, 2025-01-03:
We are closing all issues related to retro support and will be re-adding new tickets for the systems we are interested in looking at adding.

closed 2025-01-03 by TheTrainGoes