OpenStickCommunity/GP2040-CE #277: Bounty - Case RGB LED settings

TheTrainGoes, 2023-05-23:
This bounty is to add the ability to have and customize case LEDs.  

You should be able to specify either individual, a total number of or groups of RGB LEDs and be able to have them do static colors or animations that may run independently of the other LEDs.  

Example would be a board with 2x LEDs on each button that run the rainbow animation.  there would be PLEDs and then case LEDS.  The case LEDs may just stay in a single color, or do a slow change together.

fatihG, 2023-05-23:
Just to add to this. To make it 'future proof' have additional groups set by the user, or at the very least make it expendable for possible future additions. 

Mentioned briefly here https://github.com/OpenStickCommunity/GP2040-CE/issues/274
If something like this gets implement we could have separate LEDs to show what profile is being used currently.

TheTrainGoes, 2023-05-24:
@fatihG an LED to show profile is better to be included the profile request https://github.com/OpenStickCommunity/GP2040-CE/issues/274.

That will be a fair bit of work on its own.  

There is a projec tunderway to re-do how LEDs are handled that should make several of these requests easier.

TheTrainGoes, 2024-12-28:
This work has been completed.

closed 2024-12-28 by TheTrainGoes