OpenStickCommunity/GP2040-CE #286: Reset Pins only, avoid resetting all settings.

fatihG, 2023-05-26:
### Have you checked the GP2040-CE documentation [HERE](https://gp2040-ce.info/#/) to confirm this feature doesn't already exist?

Yes

### Which problem is this feature request solving?

Changing pin set up in the Web UI, no way to reset to defaults, without wiping every setting.

### Describe the solution you would like to see

Reset just the pins, not the entire device.

### Describe alternatives you've considered

Download a backup of everything, prior to making changes. 
Make changes, mess up. 
Wish to reset. 
Load backup of version prior to making changes.

### Can you submit a pull request?

Yes.

### Are you planning on working on this feature yourself?

No.

TheTrainGoes, 2023-05-29:
Hi @fatihG - this would make for a good bounty, would you be willing to co-sponsor it?

fatihG, 2023-05-29:
Ill wait for progress on this. 
https://github.com/OpenStickCommunity/GP2040-CE/issues/293

TheTrainGoes, 2023-09-06:
We are going to close this as the process for resetting is a lot easier now with Protobuff

closed 2023-09-06 by TheTrainGoes